Apollon77/ioBroker.smartmeter #376: Zähler EDL300L auslesen

wschiro, 2023-07-20:
Der EDL300L steht ja nicht in der Kompatibitätsliste. Besteht überhaupt die chance, den mit smartmeter auszulesen?
Laut Datenblatt wird SML mit 9600 Baud verwendet. 
Ich habs mit dem Adapter und diesen Einstellungen versucht, aber ich bekomme keine Verbindung. Ich habe den Lesekopf Hitchi USB verwendet.
Der raspi erkennt den Lesekopf. Wenn ich einen Dump von /dev/ttyUSB0 mache mit xxd, bekomme ich ca. alle 30 s Daten. (die ich aber nicht interpretieren kann)

Apollon77, 2023-07-24:
Wenn es SML ist dann wird das an sich gehen, die Kompatibilitätsliste ist nur so gut wie ich Rückmeldungen bekomme, also bei der Anzahl Zähler da draussen kann die nie korrekt sein.

Bite nicht in der Instanz konfig wild rumändern, sondern nur das Protokol setzen und USB port ... dann versuchen. ggf Debug log aktivieren. Support im ioBroker Forum.

closed 2023-07-24 by Apollon77

Apollon77, 2023-07-24:
Ja

wschiro, 2023-07-24:
ok, das ist das Problem, entweder der Lesekopf oder wahrscheinlicher der Zähler
danke

Apollon77, 2023-07-25:
AM besten beshcreibe doch aml im Forum was es ist und zeige die Daten und so

wschiro, 2023-07-25:
Wenn die Startsequenz in den Daten nicht vorkommt (das script von Volkszähler hat automatisiert in 2 Stunden auch keine gefunden) liegt es an meiner Hardware. Ich brauche es auch nicht dringend, weil ich auch noch mit CTs und einem DTSU über meine beiden Wechselrichter messe, allerdings nicht geeicht. In der kalten Jahreszeit baue ich mir einen Lesekopf und schaue, ob der Zähler ne Macke hat. Danke

Apollon77, 2023-07-25:
Naja oder falscher Baudrate oder anderen Einstellungen oder sowas ... Ursachen können es mehrere sein

wschiro, 2023-07-25:
Prinzipiell ja, aber wenn ich die Rohdaten von /dev/USB0 lese und der Zähler kann nur 9600,8n1, dann müsste im hexdump die Startsequenz zu sehen sein.
Es gibt ja jemanden, der mit Volkszähler erfolgreich den EDL300L ausgelesen hat und mit dem Adapter auch.

wschiro, 2023-08-15:
Ich habe jetzt einen anderen Lesekopf, mit dem ich den Zähler auslesen kann.
Mit einem Shellscript bekomme ich die Daten. Aber nicht mit dem Adapter. Alles auf Standard gelassen.
Mit dem script kann ich die Daten in eine Datei schreiben und dann versuchen, von dieser Datei mit dem Adapter zu lesen.
Wie muss diese Datei aufgebaut sein?