Apollon77/ioBroker.smartmeter #387: Update stable version in repo from 3.3.4 to 3.4.0

GermanBluefox, 2023-12-11:
# Think about update stable version to 3.4.0
**Version**: stable=**3.3.4** (639 days old) => latest=**3.4.0** (15 days old)
**Installs**: stable=**5724** (89.47%), latest=**324** (5.06%), total=**6398**

Click to use [developer portal](https://www.iobroker.dev/adapter/smartmeter/releases)
Click to [edit](https://github.com/ioBroker/ioBroker.repositories/edit/master/sources-dist-stable.json#L2326)

Note: This is an automatically generated message and not personally authored by bluefox!

mcm1957, 2023-12-25:
@Apollon77 

Any reason to delay stable update any longer?

**If there's no reaction / comment until 7.1.2024 an update will be considered anyway.**

Apollon77, 2023-12-25:
go

closed 2023-12-29 by GermanBluefox