Apollon77/ioBroker.smartmeter #400: Error while parsing SML message: Error: Unknown TL-Field 0x77

ITServ-DE, 2024-08-26:
**Describe the bug**  
Log reads:

```
Error while parsing SML message: Error: Unknown TL-Field 0x77 for OctetString [Offset: 37]!: 
Message: 1b1b1b1b01010101760500[...redacted; see personal mail ...]

```
**To Reproduce**  
Unknown

**Expected behavior**  
Probably, there is a message 

**Screenshots & Logfiles**  
If applicable, add screenshots and logfiles to help explain your problem.

**Versions:**  
 - Adapter version: 3.4.0
 - JS-Controller version: 6.0.9
 - Node version:  v18.19.0 
 - Operating system: Debian 12.6

I have sent the entire message via personal mail to `iobroker@fischer-ka.de` due to personal information in the message. If there is any possibility to assist, please let me know.

Apollon77, 2024-08-31:
Is the error the same for each request or is it just coming some times? Maybe also different Offset or Field? Or is it always and every time this same error? Please provide a full debug log.

If the error is different or not always appearing please first check cables, try an active USB hub to increase Power of protection and such. make sure the cable goes as fast as possible the most away from the smartmeter to make sure therer are not interferences. make sure the head is correctly positioned - try to move a bit in several directions with always checking results.

ITServ-DE, 2024-09-02:
I cannot see any regularity when this message occurs, so I would say thy randomly occur. 

What I can say is that 
(a) the TL-Field number changes randomly (0x77, 0x46, 0x52, 0x5d62, 0x64
(b) for every info, a warning occurs

`
2024-08-31 08:54:51.592  - info: smartmeter.0 (400111) Error: Error while parsing SML message: Error: Wrong TL-Field 0x64 for Choice!
2024-08-31 08:54:51.592  - warn: smartmeter.0 (400111) Error while parsing SML message: Error: Wrong TL-Field 0x64 for Choice!: Message
2024-08-31 09:00:58.686  - info: smartmeter.0 (400111) Error: Error while parsing SML message: Error: Wrong TL-Field (7) for Unsigned!
2024-08-31 09:00:58.686  - warn: smartmeter.0 (400111) Error while parsing SML message: Error: Wrong TL-Field (7) for Unsigned!: Message
2024-08-31 09:04:03.923  - info: smartmeter.0 (400111) Error: Error while parsing SML message: Error: Wrong TL-Field (7) for Unsigned!
2024-08-31 09:04:03.924  - warn: smartmeter.0 (400111) Error while parsing SML message: Error: Wrong TL-Field (7) for Unsigned!: Message
2024-08-31 09:15:15.231  - info: smartmeter.0 (400111) Error: Error while parsing SML message: Error: Unknown TL-Field 0x64 for OctetString [Offset
2024-08-31 09:15:15.231  - warn: smartmeter.0 (400111) Error while parsing SML message: Error: Unknown TL-Field 0x64 for OctetString [Offset: 32]!
2024-08-31 09:18:19.296  - info: smartmeter.0 (400111) Error: Error while parsing SML message: Error: Unknown TL-Field 0x52 for OctetString [Offset
2024-08-31 09:18:19.296  - warn: smartmeter.0 (400111) Error while parsing SML message: Error: Unknown TL-Field 0x52 for OctetString [Offset: 125]!
2024-08-31 09:24:25.846  - info: smartmeter.0 (400111) Error: Error while parsing SML message: Error: Wrong TL-Field (7) for Unsigned!
2024-08-31 09:24:25.847  - warn: smartmeter.0 (400111) Error while parsing SML message: Error: Wrong TL-Field (7) for Unsigned!: Message
2024-08-31 09:29:32.527  - info: smartmeter.0 (400111) Error: Error while parsing SML message: Error: Wrong TL-Field (7) for Unsigned!
2024-08-31 09:29:32.527  - warn: smartmeter.0 (400111) Error while parsing SML message: Error: Wrong TL-Field (7) for Unsigned!: Message
2024-08-31 09:39:45.471  - info: smartmeter.0 (400111) Error: Error while parsing SML message: Error: Wrong TL-Field (7) for Unsigned!
2024-08-31 09:39:45.471  - warn: smartmeter.0 (400111) Error while parsing SML message: Error: Wrong TL-Field (7) for Unsigned!: Message
2024-08-31 09:41:47.370  - info: smartmeter.0 (400111) Error: Error while parsing SML message: Error: Unknown TL-Field for SmlList!
2024-08-31 09:41:47.370  - warn: smartmeter.0 (400111) Error while parsing SML message: Error: Unknown TL-Field for SmlList!: Message
2024-08-31 09:53:01.987  - info: smartmeter.0 (400111) Error: Error while parsing SML message: Error: Unknown TL-Field for SmlListEntry!
2024-08-31 09:53:01.987  - warn: smartmeter.0 (400111) Error while parsing SML message: Error: Unknown TL-Field for SmlListEntry!: Message
2024-08-31 09:54:04.007  - info: smartmeter.0 (400111) Error: Error while parsing SML message: Error: Unknown TL-Field 0x52 for OctetString [Offset
2024-08-31 09:54:04.007  - warn: smartmeter.0 (400111) Error while parsing SML message: Error: Unknown TL-Field 0x52 for OctetString [Offset: 413]!
2024-08-31 10:01:11.487  - info: smartmeter.0 (400111) Error: Error while parsing SML message: Error: Unknown TL-Field 0x46 for OctetString [Offset
2024-08-31 10:01:11.487  - warn: smartmeter.0 (400111) Error while parsing SML message: Error: Unknown TL-Field 0x46 for OctetString [Offset: 93]!
2024-08-31 10:06:17.809  - info: smartmeter.0 (400111) Error: Error while parsing SML message: Error: Unknown TL-Field for SmlListEntry!
2024-08-31 10:06:17.809  - warn: smartmeter.0 (400111) Error while parsing SML message: Error: Unknown TL-Field for SmlListEntry!: Message
2024-08-31 10:52:08.487  - info: smartmeter.0 (400111) Error: Error while parsing SML message: Error: Unknown TL-Field 0x77 for OctetString [Offset
2024-08-31 10:52:08.487  - warn: smartmeter.0 (400111) Error while parsing SML message: Error: Unknown TL-Field 0x77 for OctetString [Offset: 125]!
`

This all looks to random for me to be a software issue, so it's likely to be more with the optical interface. I will try to move the reading head.

closed 2024-09-02 by Apollon77