Apollon77/ioBroker.smartmeter #402: efr SGM-D4-A920N

Schnietz, 2024-09-06:
Hi, 

gibt es eine Möglichkeit den "efr SGM-D4-A920N" auszulesen ? Hatte Jahrelang einen Iskra Zähler und es lief ohne Probleme. Nun werden bei uns überall die "efr" eingebaut. Leider ist er nicht gelistet und Informationen wie er ausgelesen wird sind dürftig. 

Danke fürs lesen,

Gruß, Markus

Apollon77, 2024-09-09:
Welche Einstellungen wurden genutzt=? was steht im Log? Ich hab das gefunden? passt das zu deinem Handbuch?

https://www.photovoltaikforum.com/thread/164008-z%C3%A4hler-efr-sgm-c4-ich-bekommen-ihn-nicht-zum-laufen/

gehen diese EInstellungen? Hast Du eine PIN bekommen und die freigschaltet? Oder ist es HTTP?

Ohne mehr informationen kann ich daz unichts sagen

Schnietz, 2024-09-09:
Ich war einfach zu doof und überschnell.

 Tut mir leid, läuft jetzt. Falsche Einstellung.

closed 2024-09-11 by Apollon77