Apollon77/ioBroker.smartmeter #405: "No or too long answer from Serial Device after last request" (sporadic, happens every one or two days)

G4l1Inger, 2025-01-03:
**Describe the bug**  
First let me say thank you to Apollon77 for this valuable code contribution.
I have been using this since about one year and it works very good for most of the time.
Unfortunately the polling of values stops sporadically once every one or two days.
Log always shows error message "No or too long answer from Serial Device after last request"
Windows 10 Device Manager always shows no issue with serial port COM3
In order to resume readouts I need to either pull&push the USB-connector or restart the PC

**To Reproduce**  
Steps to reproduce the behavior:
Run it and wait for more than 24 hours

**Expected behavior**  
A clear and concise description of what you expected to happen.
I would like to run it 265/24/7 without having to restart my PC or unplugging USB

**Screenshots & Logfiles**  
If applicable, add screenshots and logfiles to help explain your problem.
see attachments

**Versions:**  
 - Adapter version: v3.4.0 (no change with updates of last 12 months)
 - JS-Controller version:  v8.8.3 (no change with updates of last 12 months)
 - Node version: <node-version>  v20.18.1 (no change with updates of last 12 months)
 - Operating system: Windows 10 Pro 22H2

**Additional context**  
There is a USB-extender U2EX50 connectet in between PC and Weidmann IR-reader.
I already replaced this extender with an identical model => no change.
I already tried modifying the following COM3 settings in Windows device-manager:
- USB packet sizes reduced from 4096 to 1024 (receive and send)
- Timeouts increased from 0 to 100 (read and write timeouts)
=> no change

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Apollon77, 2025-01-03:
Sorry, but this sounds more like a topic and question for the forum ... but in fact you could do one thing to find out if data stopped being received or if garbage is received.

Whenever the issue happens again go to admin, enable expert mode and set the loglevel to debug but tick the "withoug restart" checkbox. Then check the log. is data received or is it just waiting?

But nevertheless if there is an issue with the USB port or windows or whatever then you need to find the reason there.

You could try to restart the adapter once a day automatically (scheduled restart in instance settings) and see if clsing and reopening the serial port do not show this issue ...  But adapter side I can not do anything

G4l1Inger, 2025-01-03:
OK, thanks for your advice. I'll try that.

G4l1Inger, 2025-01-04:
One more question please: 

Normally, when I unplug the USB cable with the reader, your adapter can handle it. As a result I see a gap in the graph on my flot diagram. Whenever the reader sends data again the diagram continues and flot draws a straight line from the last good value.

After unplugging the USB-cable log shows the same "warn - No or too long answer from Serial Device after last request." but additionally "info - Error: Error on Baudrate changeover: Error: Port is not open" and warn - "Error on Baudrate changeover: Error: Port is not open". But these messages are not shown in case of my problem. 

So I assume that when my problem happens the "Port" is still open and receives some corrupt data with the correct baudrate. Unfortunately I didn't have this situation again in oder to view the debug log but I am wodering how does the adapter handle corrupt data? Normally I would expect an error message or perhaps a few retry attemtps before stopping the communication.

Apollon77, 2025-01-04:
To see if corrupt or no data debug log would be needed ... see above

G4l1Inger, 2025-01-10:
OK, this took a while as I was out of house on a business trip.
When I came home I saw that smartmeter data logging has stopped again.
I checked the log data in debug mode as you said but there is no other message.
COM3 is still reported to be up and running with no issues by the devicemanager.
[image]
[image]
Upon unplugging the USB-Plug I get 3x the message "Error on baudrate changeover" and "port is not open"
Right after plugging back in the metering continues normally.
Any ideas what else I can do?

G4l1Inger, 2025-01-10:
I also tried restarting the adapter automatically once every hour.
This doesn't make any difference.

Apollon77, 2025-01-11:
Ok also the log above shows that really no data come in ... so it is not about trash data ...

And when adapter restart is not the issue then it is something with the com port or device... Sorry but I have no idea what could be the reason on windows. Sorry

closed 2025-01-11 by Apollon77

G4l1Inger, 2025-01-11:
Yes, I agree with that. Thanks for your help!